AVSLab/bsk_rl #17: Public-facing documentation

Mark2000, 2023-07-25:
At some maturity level, the project should be documented in a Read the Docs style manner. Should happen at a sweet spot between stability and complexity, maybe in 6 months or so?

johnowagon, 2023-12-06:
I noticed many of the reStructuredText style docstrings that are currently in the repo contain syntax errors. There should be a newline after the summary section of the docstring. Im thinking this can be easily done using `sed` to change the docstrings using regex, thoughts on this approach?

Mark2000, 2023-12-06:
Sounds like a good plan. Would be nice to have a way of automatically checking docstring formatting, not sure if such a tool exists

PR #140 (2024-06-12): Refactor/v1 0 0
## Description
Closes #115 
Closes #82 
Closes #80 
Closes #70
Closes #55
Closes #19
Closes #17
Closes #16
Closes #14
Closes #11
Closes #10
Closes #9
Closes #8

In progress.

### Type of change
- [ ] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [x] Breaking change (fix or feature that would cause existing functionality to not work as expected)

### How should this pull request be reviewed?
- [ ] By commit
- [x] All changes at once

## How Has This Been Tested?

Please describe the tests that you ran to ver [... cut]
closed 2024-06-12 by Mark2000