AVSLab/bsk_rl #51: Error setting up bsk_rl environment using setup.py

LorenzzoQM, 2023-09-05:
**Describe the bug**
Issues when trying to setup the environment through setup.py (pip install .) due to errors to install pyfftw as part of the requirements for chebpy. We believe this happens only on Apple M2 chip.

A workaround is to set the option "--no-deps" for chebpy in the setup.py file. 

**Expected behavior**
No errors when using setup.py.

**Version (please complete the following information):**
 - Python: [3.11.4]
-  Basilisk: [2.2.1]
 - Platform: [MacOS 13.5.1 with Apple M2 Pro chip]

Mark2000, 2023-09-12:
Opened an issue in the chebpy repo, though given that conda installs (which probably work) are preferred over pip installs for that package, it may not be resolved. Will see if any other maintainers over there have an opinion: https://github.com/chebpy/chebpy/issues/70

Mark2000, 2023-12-07:
Since this isn't being resolved by pyfftw/chebpy, @LorenzzoQM will have the `--no-deps` flag be added if the system is a arm silicon Mac.

Yume27, 2024-01-17:
note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pyfftw
Successfully built chebpy
Failed to build pyfftw
ERROR: Could not build wheels for pyfftw, which is required to install pyproject.toml-based projects

PR #110 (2024-01-18): Issue #51: Add --no-deps for chebpy in Mac ARM
## Description
Closes #51 

This pull request fixes the errors when installing pyfftw as part of the requirements for chebpy in Mac OS with ARM architecture.

### Type of change
- [X] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)

### How should this pull request be reviewed?
- [X] By commit
- [ ] All changes at once

## How Has This Been Tested?

Please describe the tests that [... cut]
closed 2024-01-18 by LorenzzoQM