AVSLab/bsk_rl #59: Battery checking should be continuous, not just at steps

Mark2000, 2023-09-13:
**Describe the bug**
The satellite can die then come back alive over the course of a step without triggering failure.

[image]

**Additional context**
Should be fixed by either logging battery and checking at the end of the step, or by adding a continuous terminal event checking battery level. I think the latter is preferable, as it will be less memory intensive.

Mark2000, 2023-12-20:
Relevant to #74 , once the battery reaches 0 the satellite should remain dead.

YYan99, 2024-10-21:
[image]
Why does the battery level increase here when it's not in charging mode?  I also meet this phenomenon when I run the example code `simple_environment`. I change the action to 'Scan', but the charge level increases sometimes:

```
while not truncated:
    observation, reward, terminated, truncated, info = env.step(action=0)
    print(f"Charge level: {observation[1]:.3f} ({env.unwrapped.simulator.sim_time:.1f} seconds)\n\tEclipse: start: {observation[2]:.1f} end: {observation[3]:.1f}")
```

YYan99, 2024-10-23:
> As in a real spacecraft, the power system models remain active even when not explicitly in a sun pointing mode. As long as there is solar incidence on the panels (i.e. positive dot product and not in eclipse), the battery will gain charge according to the incidence angle, distance from sun, and panel area. The charge mode just explicitly sets the attitude controller to point the panels at the sun.

Thanks for your detailed explanation!

closed 2026-11-04 by Yume27