AVSLab/bsk_rl #72: data_storage_valid check killing simulation when storage if full

LorenzzoQM, 2023-09-27:
Satellite data_storage_valid check killing the simulation (failed data_storage_valid check) when data buffer is equal to maximum storage capacity due to numerical rounding error (storage level slightly above storage capacity).

**Version (please complete the following information):**
 - Python: [3.11.5]
-  Basilisk: [2.2.1]
 - Platform: [MacOS 13.5.2]

Mark2000, 2023-10-17:
@LorenzzoQM reopening this since I think the behavior should be to fail if the buffer is approximately full when `storageUnitValidCheck`. Didn't catch that on the PR. Should be something like:
```
    def data_storage_valid(self) -> bool:
        """Check that the buffer has not run out of space."""
        storage_check = self.storageUnitValidCheck
        if storage_check:
            return self.storage_level < self.storageUnit.storageCapacity and not np.isclose(
                self.storage_level, self.storageUnit.storageCapacity
            )
        else:
            return True
```
with `and not` replacing what used to be `or`

Mark2000, 2023-12-05:
@LorenzzoQM can this issue be closed with your bsk changes to make storage an int?

LorenzzoQM, 2023-12-06:
Basilisk now uses [int variable type for storage](https://github.com/AVSLab/basilisk/pull/507), avoiding numerical rounding errors and fixing this issue.

closed 2023-12-06 by LorenzzoQM