AVSLab/bsk_rl #79: Refactor for src style package

Mark2000, 2023-10-31:
src directory style Python packages are preferred these days. We should be using that, with a pyproj.toml. Ensure it doesn’t break cluster install. May not be possible with how chebpy is installed.

johnowagon, 2023-11-09:
Some benefits to this structure:
1. The main benefit is that `src` style is explicit, theres no ambiguity on what is available to import. Plus I think its cleaner.
2. As of [PEP 518](https://peps.python.org/pep-0518/), using pyproject.toml is recommended anyways.

From some initial research I don't think refactoring the structure is going to cause issues during rebases, however I haven't tested this yet, I'll jump on it tomorrow.

Side note, because the setup.py script ran some code along with installation, I had to specify an entrypoint in the pyproject.toml to achieve the same functionality. This changes the install command from `pip install -e .` to `pip install -e . && finish-install`.

Mark2000, 2023-11-09:
Even though the install command would be longer, it'd probably better. I could imaging someone not liking that setup.py has calls to `sys` and downloads files from the internet with no warning; `finish-install` makes that more explicit.

We will want to test that `pyproject.toml`-based install works on the on the CURC cluster. I tried that with another small package I made for a class and couldn't get it to work due to the `build-system` specification, but didn't try very hard to find a fix to the issue. @johnowagon, @LorenzzoQM should be able to help you get an account on the cluster and get Basilisk installed there, he did it fairly recently and made some fixes to my broken cluster install script.

closed 2023-12-05 by johnowagon