AVSLab/bsk_rl #89: Ensure BSK_RL is compatible with BSK installation

Mark2000, 2023-12-01:
Relevant to #14, since our released versions should probably match certain bsk releases

I think we should check on import that a minimum version of bsk_rl is installed (by version? by commit hash for develop?) and warn the user if it is not

jbbennett91, 2023-12-05:
Any chance this is related to the value mismatch that occurred sometime last week?

Installing current versions of `basilisk` and `bsk_rl`, then testing with `bsk_rl/examples/general_satellite_tasking/single_sat.py` returns the following issue:

[image]

Mark2000, 2023-12-05:
Ah, good catch. I believe that's a result of #18, which adds a flag to the info sometimes that's not in the form `(time, message)`. I'll open a PR tomorrow to address this.

A documentation system is currently being put together and that'll change how example scripts are formatted and handled. Currently, example scripts aren't being caught in the test suite; I'll make sure that they are tested when we make that change so that PRs that break common use patterns like this are avoided in the future.

Mark2000, 2023-12-05:
@LorenzzoQM, could you review #92 which refactors the previous PR?

Mark2000, 2023-12-05:
@jbbennett91 Should be resolved now with #92 . As an additional heads up, there's a PR that'll be switching us to a src-style package coming up, so you may need to uninstall and reinstall the package.

Mark2000, 2023-12-05:
Over in Basilisk, @patkenneally is looking into automated beta version number increasing (so that each commit increases bsk version to 2.2.1b1, 2.2.1b2, 2.2.1b3, etc). This will allow us to match our develop to specific bsk versions without needing to dig around for commit hashes.

Considering code like this in the `bsk_rl/__init__.py` file, only remaining question is where `BSK_VERSION` should be specified. Needs to be in the src dir to check at import, which I think is where this check should happen given the development cycle.

```
from pkg_resources import (
    DistributionNotFound,
    VersionConflict,
    get_distribution,
    require,
)

BSK_VERSION = "2.2.1b0"
try:
    require(f"BasiliskNot>={BSK_VERSION}")
except VersionConflict:
    warn(
        f"Basilisk version {BSK_VERSION} or higher is required for full functionality. "
        f"Currently installed: {get_distribution('Basilisk').version}"
    )
except DistributionNotFound as e:
    e._template = (
        "The '{self.req}' distribution was not found "
        "and is required by {self.requirers_str}. Install from "
        "http://hanspeterschaub.info/basilisk/."
    )
    raise e
```

PR #93 (2023-12-06): Issue #89: Check Basilisk version on import
## Description
Closes #89

Checks if BSK is at least the version specified in `bsk_version_req.txt`. BSK should have beta releases labelled soon, which 

How should this pull request be reviewed?
- [ ] By commit
- [X] All changes at once

## Type of change

- [ ] Bug fix (non-breaking change which fixes an issue)
- [X] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)

## How Has This Been Tested?

Manually tested install in various configs [... cut]