AVSLab/bsk_rl #125: Requires retasking should consider the environment's maximum step duration

LorenzzoQM, 2024-03-13:
**Describe the bug**
Info for "requires_retasking" does not consider the maximum step duration of the environment to specify if the satellite needs retasking.

**Expected behavior**
The satellite should require retasking if the time elapsed from the last tasking was more than the environment's maximum step duration. 

**Version:**
 - Python: 3.11.6
-  Basilisk: 2.2.2
 - Platform: MacOS 14.3.1

Mark2000, 2024-07-24:
I'm going to take a look at this since I was likewise running into this issue. My thought is to add a flag (either to the satellite or the environment) to say whether the environment timeout should require retaking. Something like `require_retasking_on_max_step`?

At the same time, I'll address a related issue, which is that an event that is the same duration as `max_step_duration` will cause the env to stop and return `requires_retasking=False`, followed by a zero-duration step that returns `requires_retasking=True`. This is weird and potentially bad for obvious reasons.

Mark2000, 2024-07-24:
For the latter comment I made, it seems that this is because BSK makes sure that events aren't checked twice for one time. For example, if an event triggers at 10 seconds and you run the sim for 10 seconds, then 5 more second, the event will trigger only once at the beginning of the 5 second execution and not at the end of the 10 second execution.