AVSLab/bsk_rl #159: Support new RLlib API stack

Mark2000, 2024-07-13:
Keep this issue open until Ray 3.0 release. Currently identified issues:
- Callbacks have different signature
- Example scripts are for old API stack

josephTimm, 2024-08-06:
Hi, I am using bsk_rl for a research project and have been encountering problems related to this issue. What API version can I use for RLlib and it's example script to work properly in the meantime?

Mark2000, 2024-08-06:
I would probably stick with the RLlib old API stack for now, especially if you're doing single agent RL. I've found that the rapid updates to the new RLlib stack causes it to break with pretty much every release, so I'm hoping for better stability once they release 3.0. For this example script (https://avslab.github.io/bsk_rl/examples/rllib_training.html), I've used the new stack API with Ray 2.32 and I know that it breaks on 2.33. I believe the old stack should work on either, though I've only tested on 2.32.

If you're having any specific issue with the example script, happy to help debug.