AVSLab/bsk_rl #170: Having trouble installing bsk-rl

hhlei, 2024-08-08:
**Describe the bug**
When I follow the installation instructions, I'm getting this error on the pip install -e . && finish_install step: 

Traceback (most recent call last):
  File "C:\Users\henry.lei\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\henry.lei\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\henry.lei\Documents\Projects\basilisk\.venv\Scripts\finish_install.exe\__main__.py", line 4, in <module>
    from bsk_rl.finish_install import pck_install
  File "C:\Users\henry.lei\Documents\Projects\bsk_rl\src\bsk_rl\__init__.py", line 5, in <module>
    from bsk_rl.gym import ConstellationTasking, GeneralSatelliteTasking, SatelliteTasking
  File "C:\Users\henry.lei\Documents\Projects\bsk_rl\src\bsk_rl\gym.py", line 14, in <module>
    from bsk_rl.comm import CommunicationMethod, NoCommunication
  File "C:\Users\henry.lei\Documents\Projects\bsk_rl\src\bsk_rl\comm\__init__.py", line 49, in <module>
    from bsk_rl.comm.communication import (
  File "C:\Users\henry.lei\Documents\Projects\bsk_rl\src\bsk_rl\comm\communication.py", line 11, in <module>
    from bsk_rl.sim.dyn import LOSCommDynModel
  File "C:\Users\henry.lei\Documents\Projects\bsk_rl\src\bsk_rl\sim\dyn.py", line 66, in <module>
    from bsk_rl.sim import world
  File "C:\Users\henry.lei\Documents\Projects\bsk_rl\src\bsk_rl\sim\world.py", line 41, in <module>
    from bsk_rl.utils.functional import collect_default_args, default_args
  File "C:\Users\henry.lei\Documents\Projects\bsk_rl\src\bsk_rl\utils\functional.py", line 7, in <module>
    from typing import Any, Callable, ParamSpec, TypeVar
ImportError: cannot import name 'ParamSpec' from 'typing' (C:\Users\henry.lei\AppData\Local\Programs\Python\Python39\lib\typing.py)

**To Reproduce**
pip install -e . && finish_install

**Expected behavior**
Basilisk example code looks like it works as expected. bsk-rl fails on installing it into the pip environment and running finish_install.

**Version (please complete the following information):**
 - Python: 3.9.11
-  Basilisk: 2.3.0 ( I think)
 - Platform: Windows 11

**Additional context**
Add any other context about the problem here.

Mark2000, 2024-08-08:
Ah, try using Python 3.10 instead for now. It looks like we’re using some 3.10-specific typing features. I’ll look into either removing those or requiring 3.10 in the installation instructions.

hhlei, 2024-08-08:
Python 3.10 seems to have worked! However, I'm getting a different error now on the same installation step:

ModuleNotFoundError: No module named 'Basilisk.architecture.cMsgCInterfacePy'

I see from the basilisk site that this module is deprecated and to use messaging, but I'm not sure which basilisk version that refers to

Mark2000, 2024-08-08:
To my current knowledge, that shouldn’t be an issue but I’ll look into testing with the latest a Basilisk version. Did you manage to run any of the Basilisk example scripts?

Mark2000, 2024-08-08:
Odd, I’ll look into reproducing this issue later today.

Mark2000, 2024-08-08:
I am unable to reproduce `ModuleNotFoundError: No module named 'Basilisk.architecture.cMsgCInterfacePy'`. However, I created a branch that should eliminate that import: https://github.com/AVSLab/bsk_rl/pull/171

Please try checking out that branch and seeing if it works.

Mark2000, 2024-10-01:
Based on [this actions run](https://github.com/AVSLab/bsk_rl/actions/runs/11129662631/job/30927268958?pr=202), I've determined that refactoring for 3.9 compatibility is going to be too much work, as we rely on the bisect "key" feature that was introduced in 3.10. Going to drop 3.9 support instead.

PR #202 (2024-10-01): Issue #170: Remove claimed Python 3.9 compatibility
## Description
Closes #170

Python 3.9 compatibility is too hard to add, as we use the "key" key for bisect, which is not supported.

### Type of change
- [X] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)

### How should this pull request be reviewed?
- [ ] By commit
- [X] All changes at once

## How Has This Been Tested?

Remove Python 3.9 from CI.

## Future Work

None.
 [... cut]
closed 2024-10-01 by Mark2000