AVSLab/bsk_rl #224: Can we simulate without Harmonics for Gravity

mvinovivek, 2025-02-07:
Hi, 

Thank you for making this wonderful library. Learning a lot of things while using it. 

I observed that when we create an environment it by default uses the Spherical harmonics for gravity. I am wondering is it possible to run the simulation without one? 

I looked around and commented the following lines 

_1. src/bsk_rl/sim/world.py_ 

```python
self.planet.useSphericalHarmonicsGravityModel(
    bsk_path + "/supportData/LocalGravData/GGM03S.txt", 10
)
```

_2. src/bsk_rl/utils/orbital.py_
```python
planet.useSphericalHarmonicsGravityModel(
    bskPath + "/supportData/LocalGravData/GGM03S.txt", 10
)
```
But I am not sure that it is having any effect. 

I am not able to understand why the inclination changes drastically when using bsk_rl but not when using native Basilisk. Kindly see the image 

[image]

Thank you!

Mark2000, 2025-02-07:
Commenting out those lines would be my best idea on disabling spherical harmonics. I don’t think that they are your problem though, since they do not perturb an orbit that much. How are you saving the data for each plot? It appears to me that the left plot is in the inertial frame, but the right one is relative to the planet-fixed frame.

mvinovivek, 2025-02-10:
Yes. My bad. I wasn't carefully looking for the frame of references. Able to solve now. 
Commenting out the lines worked perfectly for disabling harmonics. 

Thank you!
Closing the issue.

closed 2025-02-10 by mvinovivek