AVSLab/bsk_rl #232: Error about async_multiagent_training

RainGameon, 2025-03-03:
When executing async_multiagent_training.ipynb, an error shown in Figure 2 will occur when running to the position shown in Figure 1. Could you please tell me what the cause of this error is? I'm looking forward to your reply. 
[image]

[image]

AVSlabBot, 2025-03-03:
Could you confirm that the version of Ray you’re using is the same as the one listed in the online version of the documentation?

RainGameon, 2025-03-03:
> Could you confirm that the version of Ray you’re using is the same as the one listed in the online version of the documentation?

Thank you very much for your reply. The version of Ray is '2.35.0'.

Mark2000, 2025-03-03:
Ah, did you manually apply this patch to your Ray installation: https://github.com/ray-project/ray/pull/46721

It's fixed in newer version of Ray, but we don't support other issues that arise from upgrading to those version yet.

RainGameon, 2025-03-04:
> Ah, did you manually apply this patch to your Ray installation: [ray-project/ray#46721](https://github.com/ray-project/ray/pull/46721)
> 
> It's fixed in newer version of Ray, but we don't support other issues that arise from upgrading to those version yet.

Your reply is extremely helpful. Thank you very much.

RainGameon, 2025-03-07:
> 啊，你是否手动将此补丁应用到你的 Ray 安装中：[ray-project/ray#46721](https://github.com/ray-project/ray/pull/46721)
> 
> 该问题已在 Ray 的较新版本中修复，但我们尚不支持升级到该版本后出现的其他问题。

For MARL, what should be modified if you want to use a custom policy? May I ask why no information can be printed in functions such as `init` of `PPOTorchPolicy`? I'm sorry for the interruption. Thank you for your reply.

PS: 
```
config.multi_agent(
    policies={"p0"},   
    policy_mapping_fn=lambda *args, **kwargs: "p0",
).rl_module(
    model_config_dict={
        "use_lstm": False,
        "fcnet_hiddens": [2048, 2048],
        "vf_share_layers": False,
    },
    rl_module_spec=MultiRLModuleSpec(
        module_specs={
            "p0": RLModuleSpec(),
        }
    ),
)
```

Mark2000, 2025-03-07:
I’ve not modified that part of RLlib before and that’s outside the scope of this repo. Check out their documentation for some starting points: https://docs.ray.io/en/master/rllib/index.html#customizing-rllib

closed 2025-03-07 by Mark2000