AVSLab/bsk_rl #234: Fine-Grained Attitude Control in bsk-rl for Satellite Imaging

MingJunF, 2025-03-05:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Currently, agents in Basilisk can only select high-level targets, and the satellite autonomously rotates to align with them. (correct me if I am wrong)
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Add an option for low-level attitude control, allowing the agent to:
Directly command attitude setpoints (MRP etc).
Control reaction wheel torques instead of using automated pointing.
Switch between high-level and fine-grained control modes for flexibility.
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
Overriding RWControlTask to accept direct torque commands from RL action(e.g [0.003,0.0023,0.04]).
**Additional context**
Add any other context or screenshots about the feature request here.
Fine-grained control would enhance RL&MARL research. Would the team consider this feature? I’d be happy to contribute or discuss further!

Mark2000, 2025-03-07:
I'm currently working on a branch (`feature/relative-motion`) that supports continuous actions generally, among other things. I plan to start pulling things from there into develop in the next month or two, with the continuous action framework being the first thing to be added in; however, feel free to look at that branch, but there is no guarantee of stability until it's pulled into develop.

Mark2000, 2025-04-09:
@MingJunF this PR should be merged soon, will add continuous action support: https://github.com/AVSLab/bsk_rl/pull/247

MingJunF, 2025-04-09:
Thanks, Mark—that's great to hear! I'm looking forward to the continuous action support being merged. Appreciate the update!