AVSLab/bsk_rl #279: Simulation won't termiante if time limit is not multiple of simulation time step

LorenzzoQM, 2025-06-12:
**Describe the bug**
The simulation will get stuck close to the time limit if the `time_limit` is not an exact multiple of the simulation time step (`sim_rate`). It seems that Basilisk will stop propagating to avoid exceeding the time limit while BSK-RL condition checks if `sim_time >= time_limit`. 

**To Reproduce**
Steps to reproduce the behavior:
1. Set `time_limit=10.25`.
2.  Set `sim_rate=0.5`
3. Step through the environment.

**Expected behavior**
The simulation should terminate/truncate instead of staying in a loop.

**Version:**
 - Python: 3.11
-  Basilisk: 2.7.13
 - Platform: macOS 15.5

Mark2000, 2025-06-12:
It may be beneficial to address #269 when looking into this issue, since the current code that handles this is really hard to debug.

Mark2000, 2025-07-01:
@LorenzzoQM I think either we round the time_limit up or down, or we check truncated accounting for +sim_rate. I think rounding up is the best option, because then if the user expects something to happen at the time_limit, that time will occur.

LorenzzoQM, 2025-07-01:
@Mark2000 I agree that rounding up the simulation time being passed to Basilisk is the best option!

PR #292 (2025-07-01): Bugfix/event issues
## Description
Closes #269
Closes #291 
Closes #279
Closes #287

Fixes a variety of event system issues.

### Type of change
- [X] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)

### How should this pull request be reviewed?
- [X] By commit
- [ ] All changes at once

## How Has This Been Tested?

*Please describe how tests have been updated to verify your changes.*

## Future Work

#287 fix is not robust. [... cut]
closed 2025-07-01 by Mark2000